Georgia-Tech-Off-Road/DAQ-DEPRECATED #2: 3D print brake light battery and radio holder

P7R7M, 2023-01-11:


Ian-Boraks, 2023-01-13:
Finished printing v1 on Jan 12

Ian-Boraks, 2023-01-15:
V1 had a good fit, but needs to be increased in size and tabs added

Ian-Boraks, 2023-01-15:
https://github.com/Georgia-Tech-Off-Road/DAQ/issues/11

Ian-Boraks, 2023-01-15:
Archived #11 as this has now merged the two. Today we added new tabs and a made the box bigger. Param will be printing it latter in the week for a V2 test

Ian-Boraks, 2023-01-15:
https://github.com/Georgia-Tech-Off-Road/DAQ/commit/43b6f6a713b48867f5e4790277460cfd42a0ae8f

Ian-Boraks, 2023-01-23:
Fit on car was "ok". Need to move the holes to the other side and add a grommet system.

Esittt, 2023-01-25:
https://github.com/Georgia-Tech-Off-Road/DAQ/commit/6d64d8a79a257bd8de2b01a2ad8e14167b328918

Ian-Boraks, 2023-01-27:
Waiting until new car is built to do final measuring for tab spacing

Ian-Boraks, 2023-04-13:
Plate for mounting is being manufactured

closed 2023-04-13 by Ian-Boraks