Georgia-Tech-Off-Road/DAQ-DEPRECATED #5: Create wireless driver communications PCB's

P7R7M, 2023-01-11:


nikola-rogers, 2023-01-18:
Created PCB board that will go in the car and added teensy and xbee to the board. Still need to choose LCD/OLED screen and add it to the board. Still need to create the board that will be used by the crew to send messages to the driver.

nikola-rogers, 2023-01-20:
Edited sketch for driver board and created accompanying PCB. Param and I also found the LCD we want to use and put it in the purchasing spreadsheet.

nikola-rogers, 2023-01-22:
We changed from having a separate pcb board for the lcd screen in the car to simply having that lcd connect to the teensy in the dashboard (which is connected to an xbee). The old board I created (driver screen board) will now be used to send messages to the driver using switches, and I already updated the sketch. Will create the board file at the next meeting.

nikola-rogers, 2023-01-25:
Updated the schematic and board again today.

nikola-rogers, 2023-01-27:
Finalized the schematic and board, it is ready to be printed. Started the code that will be used to send the messages. Will have to figure out how to send data on which switches are turned on from teensy to xbee at the next meeting.

nikola-rogers, 2023-02-01:
Finalized the code that will be used to send messages. Now, we will have to order the pcb and connect everything, and also buy and connect everything to the dashboard pcb.

nikola-rogers, 2023-02-03:
Created a file for the code that will be run on the dashboard to include receiving messages. Will have to finish writing at the next meeting.

nikola-rogers, 2023-02-05:
Finalized the code that will be used on the dashboard to receive and display messages. Will now move to testing.

nikola-rogers, 2023-02-10:
Tested the code to send data and got it to work. Next time: test receiver code; and ensure its correct and change sender to only send a value when the switches change and not continuously send data.

nikola-rogers, 2023-02-17:
Updated so that it only sends when the switches change, not always sending a value to dashboard. Next time: test receiver code.

nikola-rogers, 2023-02-22:
Corrected code used to receive data, still need to test.

nikola-rogers, 2023-03-03:
Param and I completely tested the sender and receiver code, both fully work!