Georgia-Tech-Off-Road/DAQ-DEPRECATED #7: Strain Gauge Validation

P7R7M, 2023-01-11:


Vishnav03, 2023-01-25:
Need to correct code that convert voltage to strain.

Vishnav03, 2023-01-27:
Changed strain gauge library to calculate actual strain, issue before was that we were using the amplified voltage to make strain measurements, but we need the actual voltage difference across the Wheatstone bridge to make strain calcs.

Vishnav03, 2023-02-01:
Got strain gauge to glue to aluminum component, need to glue solder pads and then solder wires to strain gauge.

P7R7M, 2023-02-03:
need to chck and fix code

Vishnav03, 2023-02-10:
Code fixed, strain gauge wasn't mounted properly

P7R7M, 2023-02-10:
Strain gauge wires have a nonconductive coating on them that doesn't melt with high temperature, not sure how to trim it away or if it's meant to not be trimmed away, so I mounted it without trimming and confirmed with a multimeter that the resistance across the pads read 350 ohms, now just need to actually test with code again