Georgia-Tech-Off-Road/DAQ-DEPRECATED #8: Organize powerchest w/ documentation

P7R7M, 2023-01-11:


Ian-Boraks, 2023-01-23:
@P7R7M What type of documentation are you looking for? @Esittt and myself have organized the chest.

[image]

P7R7M, 2023-01-23:
@Ian-Boraks I'm thinking if we had a document just explaing the parts that go into it (both mechanically and electrically), and explanation/diagram of how everything is connected/organized. On tuesday I can look into what we have already and lyk

Ian-Boraks, 2023-01-23:
> @Ian-Boraks I'm thinking if we had a document just explaing the parts that go into it (both mechanically and electrically), and explanation/diagram of how everything is connected/organized. On tuesday I can look into what we have already and lyk

Sounds good 👍🏻

Ian-Boraks, 2023-01-27:
https://github.com/Georgia-Tech-Off-Road/GTORDocumentation/commit/c188721a480932202cb3e8a11490d802b5a6d029

Ian-Boraks, 2023-01-27:
Fin

closed 2023-01-27 by Ian-Boraks