Georgia-Tech-Off-Road/DAQ-DEPRECATED #16: CVT Belt Temperature Measurement

Vishnav03, 2023-02-01:


Esittt, 2023-02-01:
https://dirtmonkeyperformance.com/products/aim-sports-cvt-belt-temp-system
- Potential CVT Belt Temperature Sensor
- Obvious con of being $350

Esittt, 2023-02-01:
https://www.rockymountainatvmc.com/parts/trail-tech-tto-cvt-belt-temp-sensor-p
- Cheaper alternative to the one linked above
- Two cons, it has a small display so might be hard to read at high speeds as well as it unfortunately causes cancer and reproductive harm.

Esittt, 2023-02-01:
https://digitalcommons.calpoly.edu/mesp/501/
- ECVT Design @ Calpoly

Esittt, 2023-02-12:
https://www.baseapp.com/nodesense/wireless-temperature-sensors/#:~:text=This%20sensor%20consists%20of%20photodetectors,generated%20provides%20an%20accurate%20reading.
- Wireless Temp Sensor article, potentially can help with us connecting the temp sensor to the Aux Daq.

Vishnav03, 2023-02-15:
https://www.analog.com/media/en/technical-documentation/data-sheets/tmp35_36_37.pdf

Esittt, 2023-02-17:
Look into the numbers, look at the data sheet. Look how it would work inside the CVT case. 100 degrees Celsius. Look how to repurpose a covid temperature scanner.

CVT Primary Belt runs at 240 F or 115.5 C
CVT Case / Ambient Environment runs at 120 F or 48.88 C

Esittt, 2023-02-17:
Potential Infrared Scanner:
https://www.omega.com/en-us/temperature-measurement/noncontact-temperature-measurement/handheld-infrared-industrial-thermometers/p/OS880-Series

Esittt, 2023-02-17:
Infrared Temp Sensor for CVT Belt:
https://ruggedroutes.com/UTV-belt-temperature-gauge-sensor-lowrance-off-road-gps.html

Ask the mechanical team about if we can orientate this temp sensor into the case, and drill holes on the side to put the wires through it.

Esittt, 2023-02-22:
We are able to mount this sensor in the case. Currently checking the rulebook on CVT case mounting, we just need to figure out how we will orientate the wires.

Esittt, 2023-02-22:
[image]

Esittt, 2023-02-22:
[image]

Vishnav03, 2023-02-25:
https://www.omega.com/en-us/temperature-measurement/noncontact-temperature-measurement/fixed-infrared-temperature-sensors/p/OS210-150-300-800

Vishnav03, 2023-02-26:
Thermopile Sensor: https://www.digikey.com/en/products/detail/te-connectivity-measurement-specialties/G-TPMO-104/9842001

Vishnav03, 2023-03-03:
Schematic File for Thermopile: https://www.mouser.com/ProductDetail/Measurement-Specialties/TSD305-1SL10?qs=l7cgNqFNU1hQbECGfV0ZYg%3D%3D

Ian-Boraks, 2023-03-12:
Added in https://github.com/TEConnectivity/TSD305_Arduino_Library @ https://github.com/Georgia-Tech-Off-Road/GTORHardwareLibraries/commit/5ab0bcab7a6f0289de58731ee04b827f2848fa8c . Needs testing to see if it works. Pushed to Main YOLO

Vishnav03, 2023-03-30:
Recieve PCB and Sensor. Andrew is working on cadding mount for sensor, Evan is trying to test if the code for the sensor works

Vishnav03, 2023-04-16:
CAD done, Solder done, need to get the code working